picosonic/bbc-fdc #4: Add some common disk image formats to output types

picosonic, 2018-02-07:
Currently, I've only added support for **.RAW**, **.SSD** and **.DSD** outputs.
The **.SSD** and **.DSD** files work fine in emulators except they don't support copy protected disks very well and also lots of potentially valuable information is not stored.

Some potential other disk image formats I _could_ support are : 

* **.DSK** (libdisk) disk image file format
* **.DFI** DiscFerret
* Kryoflux STREAM
* **.IMD** ImageDisk
* **.HFE** HxC Floppy Emulator
* **.SCP** Supercard Pro
* **.IPF** (Interchangeable Preservation Format) Software Preservation Society
* **.FSD** Bill Carr's FSD BBC Micro disc image
* **.FDI** Formatted Disk Image
* **.UEF** Unified emulator format
* **.ADS** / **.ADM** / **.ADL** e.t.c ADFS disk images

Alternatively I could create my own format to store both parsed data and raw sample data (as RLE).

picosonic, 2018-02-09:
Added basic .FSD output support # #

picosonic, 2018-03-20:
Added basic **.DFI** - "DiscFerrect flux dump" DFE2 output support

tkurbad, 2018-10-25:
Hi @picosonic,

awesome project! I tried to accomplish sth. similar with the proprietory Kryoflux board (s. https://github.com/tkurbad/kryofluxer) to sample my large collection of Amiga floppies before they've become completely deteriorated...

However, your solution would be much more "native" to the RPi than mine.

To come to the question of image formats, I would be very interested in the following:
* **.IPF**
* **.HFE** (version 3, if possible)
* **.ADF**

On a side note: Can your board (i.e. the raw circuit board, w/o components) be bought somewhere?

Thanks in advance!
Torsten

tkurbad, 2018-10-25:
Hi @picosonic,

> The .ADF file I had listed was for ADFS format floppies, rather than amiga.
> But it sounds like it's just a sector dump of the entire disk. Which my code can do at present when using the .IMG output format. You'd just need to rename it to .ADF. Although this wouldn't work with many copy-protected discs with non-standard formatting.

Yes, to my understanding, an .adf is a raw sector dump. The copy-protection problem was the reason for asking about HFEv3 support, which preserves the physical structure much more closely.

In the end, I want to be able to use all my Amiga disks with the Gotek Floppy emulator I bought, which is equipped with @keirf's phantastic FlashFloppy firmware.

> You may find this page interesting ..
> http://amiga.robsmithdev.co.uk/history

Looks very interesting indeed. However, I'm not THAT deep into the matter, yet.
Nonetheless, I probably should be, especially because the possibilities of formatting / copy protecting Amiga disks seem virtually unlimited.
The solution I'm using right now is struggling very hard to distinguish between intentionally "broken", i.e. copy protected, and *really* broken floppy disks. The fact that all of my floppies are at least 25 years old, doesn't make this any easier.

> I got a run of 10 of my PCBs made and have one unpopulated board left if you're interested in it.

Yes, I definitely am! All the components are in my drawer. I even got some SN7406N left over from a previous project.
Please name the amount of money I'd have to transfer to you so you'd be comfortably parting from the last board and sending it to Germany :-).

Best,
Torsten

picosonic, 2019-02-11:
Added basic **.TD0** output support

sdsnatcher, 2019-10-04:
I would like to suggest some more disk image formats:

- D88: NEC PC-88 disk image
- DMK: [Protected disk image (TRS-80 and MSX](https://github.com/lutris/openmsx/blob/master/doc/DMK-Format-Details.txt)
- XDF: Sharp X68000 disk image

picosonic, 2019-10-04:
Hi @sdsnatcher 

Are you able to provide some sample .RFI files which have been captured from the disk formats you are interested in.

I can then take a look at the data and work out the transcoding process involved.

Thanks

Jasper

sdsnatcher, 2019-10-04:
I still have to build the interface, so this might take a while.    :)

picosonic, 2019-10-16:
I was given an Amiga disk to image, and have managed to extract data.

I've added basic **.ADF** output support for Amiga formatted disks.

@tkurbad - Something you may be interested in.

drdpj, 2020-07-30:
SCP would play nicely with most extant tool chains. IPF is probably a can of worms you don't want to get into! Keir Fraser's disk analyser can convert to that for amiga format floppies if necessary, as well as many other formats, and the HxC floppy emulator software does mop up the gaps quite effectively.

CommanderCoder, 2020-09-28:
I have a lot of old Atari ST disks and I wonder if the BBC-FDC could capture AtariST format disks.  It looks like the popular image formats are:  .MSA , .ST  and maybe:  .DSK

I also have some TRS-80 disks (made on EACA Video Genie) which I haven't tried out yet, but I will.

picosonic, 2020-09-28:
I don't see why not. It captures raw flux data from the disk in the same way the FDC chip would get it. 

The extraction of sectors to disk images shouldn't be too difficult.
The AtariST low level format is documented in great detail [here](https://info-coach.fr/atari/software/FD-Soft.php)

rataweb, 2021-03-16:
Hi,
I have old 2HD floppy disk formatted to 1.7MB (DMF). Please, could you add this feature (read tracks 81 and 82)?  
Thanks,