picosonic/bbc-fdc #14: Add Apple GCR Support

philtimmes, 2020-01-21:
Apple GCR support should be added... 
It is similar enough to Commodore GCR, except that the sector count does not change and the preamble and sector headers are different.

picosonic, 2020-01-22:
@philtimmes I've been trying to do this already, but I don't have any Apple formatted floppies with which to use as samples.

philtimmes, 2020-01-22:
I can send you some.

philtimmes, 2020-01-22:
I can also send some rfi captures.

picosonic, 2020-01-22:
Sure. Can you send me a link to somewhere I can download some .rfi captures? Thanks

philtimmes, 2020-01-22:
I'm going to make some today... I have to build another BBCFDC

picosonic, 2020-02-17:
Have you been able to build a BBCFDC and get some captures yet?

philtimmes, 2020-02-27:
Nothing from china yet... seems coronavirus is having an effect on us as well...

philtimmes, 2020-05-31:
Should I close this, and re-open when I'm able to send some images?

picosonic, 2020-06-01:
No worries, I can get to it when you have something available for me to look at.

I did try processing some AppleSauce .a2r and .woz images but the raw flux formatting is quite different so it would be good to see data captured as .rfi format

picosonic, 2020-07-13:
I've obtained some Apple II 5.25" floppy disks today, so will have a go at imaging them

picosonic, 2022-01-07:
Hi, just following up on this, were you able to make some RFI images?