picosonic/bbc-fdc #15: PCBs - not an issue, more of a personal message.

JasonStonier, 2020-07-14:
Hi Picosonic - loving the project, and I just ordered some PCBs made so I can build and play with one myself. It was the same price for me to buy 10 PCBs as for 1 so if you have any requests for bare PCBs I have 9 spare for the cost of the postage. Cheers.

picosonic, 2020-07-15:
Hi,

  Thanks for the kind words. I know there are a few people who are still interested in boards

You could put a note on stardot forums about having some spares available ..

  https://stardot.org.uk/forums/

I know somebody on twitter wanted one too @ajgbarnes

Cheers

Jasper

ajgbarnes, 2020-07-15:
Thanks @picosonic - yes please @JasonStonier I would happily take one or two!

JasonStonier, 2020-07-15:
Cool - will do. Didn't want to do that without your permission as I didn't want it to look like I was profiting from your excellent work. While I'm populating one board I may as well populate 10 so I could potentially build up a set of 10 functional boards for sale at cost to the community.

ajgbarnes, 2020-07-15:
My preference is for a populated board as my soldering skills aren't great :)

JasonStonier, 2020-07-15:
@ajgbarnes - no worries. My boards are on their way to me from China now - be a couple weeks or so before they arrive, Do you want one bare or populated? Assuming I don't get done for import tax, a bare board would be £1 plus postage, a populated board maybe £5 plus postage - haven't actually costed out the components yet.

JasonStonier, 2020-07-15:
@ajgbarnes replies crossed over - I'll build you up a board. You in the UK?

ajgbarnes, 2020-07-15:
Yes in the UK and not going anywhere until this coronavirus crisis ends :)

kwoot, 2020-07-15:
Although I already have parts for 5 I lack the time, and very much the skills to build a reliable board. If I can make a trade with Jos for one working board and he gets all my parts (no pun intended) I'd be happy to help out.

picosonic, 2020-07-15:
I've got a bag of spare 34-way IDC connectors for connecting the floppy cable to the PCB, which I'd be happy to send you for no cost

JasonStonier, 2020-07-15:
Sounds like a plan is coming together! @picosonic - Jasper - presume you're in the UK as well?

JasonStonier, 2020-07-15:
> (based on the assumption that Jos is living in the Netherlands as his name suggests)

Jos?

Ik kan een beetje Nederlands begrijpen, maar ik ben in het UK...

JasonStonier, 2020-07-15:
Jasper, maybe we could chat by email to make a plan with the IDCs when you have time - jason.stonier@gmail.com

kwoot, 2020-07-15:
These Dutchies are really everywhere :-). Okay. I can still send my stuff over if you are willing to supply me with a working board?

JasonStonier, 2020-07-15:
No problem kwoot. Happy to help out. Maybe drop me an email and we'll make a plan. You can also help me improve my Dutch ;-)

JasonStonier, 2020-07-17:
[BOM.xlsx](https://github.com/picosonic/bbc-fdc/files/4937343/BOM.xlsx)

Attached a quick pricing sheet with links to the components if anyone finds it useful. I can't upload to the main branch without forking it, and I don't really want to do that.

picosonic, 2020-09-16:
I'm closing this thread.

closed 2020-09-16 by picosonic