picosonic/bbc-fdc #18: hfe support

PeterBBCUser, 2020-08-10:
Is it possible to support hfe file types as that format is more useful than fsd, especial if you have a flashfloppy drive emulator and for any type of copy protected discs that you may want to copy with it, as fsd's are not supported by flashfloppy
Thanks,
Peter.

picosonic, 2020-08-10:
I’m currently working on adding .SCP SuperCard Pro write support. But will certainly consider other formats if there is interest.

There may be existing tools to convert between .SCP and other raw formats.

PeterBBCUser, 2020-08-11:
yes HxCFloppyEmulator will let you save it in other formats and it does support .SCP (and many other formats) which meant I will then be able to export it as .HFE with that tool.
Thanks.

picosonic, 2020-08-11:
I'll close this issue for now, as it's already covered by issue #4

closed 2020-08-11 by picosonic