picosonic/bbc-fdc #23: non standard .scp .fsd format

PeterBBCUser, 2020-09-16:
Created a .scp file from a BBC disc but it will not load into HxCFloppyEmulator it just sits there saying loading and does not go further.
so suspect the .scp file that bbcfdc creates is a non standard format.
creating a .ssd works fine so the hardware and floppy drive work fine with bbcfdc, just not with trying to create protected disc images.
Also tried the .fsd format but I was unable to load the image in beebjit which can read .fsd images.

picosonic, 2020-09-16:
Yeah, the .SCP format is still in development and not in a useable state yet.

I'll post on the README when it's ready for use.

I've never had a problem with .FSD images, do you get an error from beebjit that I can investigate further? Are you able to share the .FSD with me so I can take a look?

PeterBBCUser, 2020-09-16:
its a protected disc "doctor who" that I tried to copy and beebjit comes up with BAILING: fsd sectors really cannot fit.
sure would love to share it with you don't see a way of doing that hear.
have posted an image of it on BBC Forum [https://stardot.org.uk/forums/viewtopic.php?f=8&t=20088&p=288436#p288436](url)

picosonic, 2020-09-16:
Thanks for the file. The .FSD file is valid in that it meets the spec for .FSD files.

The error you are seeing relates to beebjit finding more sectors with data than it thinks should be able to be stored on a single disk track, so it thinks there is a data overflow and "bails" because it's never seen this before and doesn't know how to handle it.

Looking at the file you submitted several of the tracks report having 20 sectors which appears to be because several of the sector ids are duplicated, probably because they come from separate sides of the disk.

I believe .FSD files are essentially single sided in nature and you may have imaged both sides of the disk into the FSD (which beebjit doesn't like). Can you try converting to FSD again but with only a single side being processed.

For example :

./bbcfdc -o test.fsd -ss

PeterBBCUser, 2020-09-16:
Okay my bad I had not used the -ss switch to create a .fsd image, have now done that and it appears to work fine with the disc I was testing, sorry about that, I will close this ticket/issue and will wait for support of .scp to be completed as you said it was still work in progress.

closed 2020-09-16 by PeterBBCUser

picosonic, 2020-09-16:
Thanks for the update