picosonic/bbc-fdc #24: double step

PeterBBCUser, 2020-09-18:
option to read 40 track disc's in a 80 track drive, i.e. a double step (40/80) switch.

picosonic, 2020-09-18:
The facility to do double stepping is automatic so long as it can detect data on the tracks.

It does this as follows ..

Seek to track 2
Read sector headers
If IDAM has track 2 then we know stepping is correct for this disk and drive
If IDAM has track 1 then we know it's a 40 track disk in an 80 track drive

Once it has established double stepping is needed it prints the message :

"40 track disk detected in 80 track drive, enabled double stepping"

Alternatively if track 2 has sectors with IDAM using track 4 then we know it's an 80 track disk in a 40 track drive, or the drive is an 80 track with a selector switch set to 40 track. When it detects this it prints the message :

"80 track disk detected in 40 track drive
*** Unable to fully image this disk in this drive ***"

I haven't created a command line option to override this behaviour as I felt it might complicate things for the user as you would also need to specify the drive tracks and disk tracks plus it could damage some drives if you double step when you don't need to as it would attempt to seek too far.

PeterBBCUser, 2020-09-18:
hmm okay let me try a different 40 track disc as the one i did try not work.
And will have to check more closely to see what messages I get back from the program.

picosonic, 2020-09-18:
I made a change today to the logic for this to add to the output if it thinks the disk may be copy protected or that the track id in the IDAM is unexpected (not one of 1/2/4).

So you may want to update your source code and recompile.

PeterBBCUser, 2020-09-18:
the 80 track side of the disk works fine the 40 track side still has problems even with the latest version (reading 40 track disk in a 80 track drive).
have attached a copy of the fsd of Repton3 40 track.

[Repton3-40t.zip](https://github.com/picosonic/bbc-fdc/files/5247434/Repton3-40t.zip)
Another think I noticed if you have a flippy disc or I am sure with a disk 40 trk one side and 80 on the other it reads the disk format from drive 0 and then processed to copy the other side with the incorrect format.

picosonic, 2020-09-19:
In your attached FSD image it has detected 40 track in an 80 track drive and stepped correctly, however track 1 has both sides stored in it which beebjit doesn't like. The other tracks seem to have the contents from every other track.

e.g.

"physical = logical"
0 = 0
1 = 2
2 = 4
3 = 6
4 = 8
e.t.c

This would only normally happen when imaging an 80 track disk in a 40 track drive (or an 80 track drive set to double step) or when double stepping was enabled in software when it shouldn't be.

Track 1 is used as the auto discovery track to determine if a disk is dual sided, did you specify single sided?

Since Bill Carr confirmed that FSD is meant to be single sided only, you'd need to capture each side separately if you want to save the disk as FSD format.

The ability to detect and correctly process both 40 and 80 track on separate sides of the same disk is down in issue #1 however this would only help when imaging to formats storing both sides of the disk.

PeterBBCUser, 2020-09-20:
this image was made with the -ss switch (as you had said previously fsd is singe sided so have been testing all the images with the -ss switch since then) which is why I don't understand why its doing odd things, and hence the reason for reporting it.

picosonic, 2020-09-20:
I've made improvements to the handling of single sided captures when outputting to FSD and TD0.

Tracks written to image can no longer contain sectors from both sides of the disk.

If you are still getting unusual results, please can you capture to RFI format and send the file.

PeterBBCUser, 2020-09-21:
still seems to trip up on track 1for some reason FSD Report gives the following on the created fsd
[image]
Attached is a rfi of the 40 track side of the Repton3 disc
[Repton3-40tss0.zip](https://github.com/picosonic/bbc-fdc/files/5257518/Repton3-40tss0.zip)

picosonic, 2020-09-22:
The RFI file looks fine, it contains 40 tracks (0..39), all are captured from side 0
Correct drive stepping for this disk and drive

`RFI
{date:"21/09/2020",time:"18:12:36",tracks:80,sides:1,rate:12500000,writeable:1}
{track:0,side:0,rpm:300.04,enc:"rle",len:203424}
{track:1,side:0,rpm:300.06,enc:"rle",len:223949}
{track:2,side:0,rpm:300.08,enc:"rle",len:205279}
{track:3,side:0,rpm:300.10,enc:"rle",len:204208}
{track:4,side:0,rpm:300.11,enc:"rle",len:204585}
{track:5,side:0,rpm:300.13,enc:"rle",len:201912}
{track:6,side:0,rpm:300.14,enc:"rle",len:210220}
{track:7,side:0,rpm:300.16,enc:"rle",len:210693}
{track:8,side:0,rpm:300.17,enc:"rle",len:197702}
{track:9,side:0,rpm:300.18,enc:"rle",len:186874}
{track:10,side:0,rpm:300.19,enc:"rle",len:185202}
{track:11,side:0,rpm:300.20,enc:"rle",len:225628}
{track:12,side:0,rpm:300.21,enc:"rle",len:218072}
{track:13,side:0,rpm:300.22,enc:"rle",len:216371} 
{track:14,side:0,rpm:300.23,enc:"rle",len:213316}
{track:15,side:0,rpm:300.24,enc:"rle",len:216366}
{track:16,side:0,rpm:300.25,enc:"rle",len:214192} 
{track:17,side:0,rpm:300.26,enc:"rle",len:203448}
{track:18,side:0,rpm:300.27,enc:"rle",len:199994}
{track:19,side:0,rpm:300.27,enc:"rle",len:207007}
{track:20,side:0,rpm:300.28,enc:"rle",len:212505} 
{track:21,side:0,rpm:300.29,enc:"rle",len:197910}
{track:22,side:0,rpm:300.30,enc:"rle",len:210809}
{track:23,side:0,rpm:300.30,enc:"rle",len:196273}
{track:24,side:0,rpm:300.31,enc:"rle",len:213210}
{track:25,side:0,rpm:300.32,enc:"rle",len:195466}
{track:26,side:0,rpm:300.32,enc:"rle",len:206131}
{track:27,side:0,rpm:300.33,enc:"rle",len:195326}
{track:28,side:0,rpm:300.34,enc:"rle",len:211532}
{track:29,side:0,rpm:300.35,enc:"rle",len:202657}
{track:30,side:0,rpm:300.35,enc:"rle",len:214798}
{track:31,side:0,rpm:300.35,enc:"rle",len:197876}
{track:32,side:0,rpm:300.36,enc:"rle",len:205992}
{track:33,side:0,rpm:300.37,enc:"rle",len:214922}
{track:34,side:0,rpm:300.37,enc:"rle",len:212736}
{track:35,side:0,rpm:300.38,enc:"rle",len:212004}
{track:36,side:0,rpm:300.38,enc:"rle",len:197895}
{track:37,side:0,rpm:300.38,enc:"rle",len:210866}
{track:38,side:0,rpm:300.38,enc:"rle",len:195986}
{track:39,side:0,rpm:300.38,enc:"rle",len:196150}`

The catalogue looks ok

`Detected DFS, side : 0
Disk title : "REPTON 3"
Disk size : 80 tracks (800 sectors, 204800 bytes)
Boot option: 3 *EXEC !BOOT
Write operations made to disk : 21
Catalogue entries : 17
D.REPTON2 000970 000970 000090 13f L
FINALE    0031e0 000000 002620 118 L
TOCCATA   0031e0 000000 002620 0f1 L
PRELUDE   0031e0 000000 002620 0ca L
MV        001900 ff8023 0001fd 0c8 L
MASTER    000e10 000e10 0002b0 0c5 L
E.REPTON3 000400 000400 000400 0c1 L
E.REPTON2 001200 002dd0 004600 07b L
E.REPTON1 002f00 002f00 002900 052 L
E.EDITOR  ff0e00 ff802b 0000e9 051 L
D.REPTON4 005810 006000 000860 048 L
D.REPTON3 000400 000400 000400 044 L
D.REPTON1 ff0e00 ff802b 000153 042 L
SPEAKSC   004700 005700 003900 009 L
FONTS     000c00 000c00 000100 008 L
MENU      001900 ff8023 00043d 003 L
[... 554 lines cut ...]
    C38 H0 S2 N1 (1) code 20
    C38 H0 S3 N1 (1) code 20
    C38 H0 S4 N1 (1) code 20
    C38 H0 S5 N1 (1) code 20
    C38 H0 S6 N1 (1) code 20
    C38 H0 S7 N1 (1) code 20
    C38 H0 S8 N1 (1) code 20
    C38 H0 S9 N1 (1) code 20

  Track : 39 (39)(ff readable)
  Sectors: 10
    C39 H0 S0 N1 (1) code 20
    C39 H0 S1 N1 (1) code 20
    C39 H0 S2 N1 (1) code 20
    C39 H0 S3 N1 (1) code 20
    C39 H0 S4 N1 (1) code 20
    C39 H0 S5 N1 (1) code 20
    C39 H0 S6 N1 (1) code 20
    C39 H0 S7 N1 (1) code 20
    C39 H0 S8 N1 (1) code 20
    C39 H0 S9 N1 (1) code 20`

Attached is the FSD output - which runs fine in beebjit

[rept.zip](https://github.com/picosonic/bbc-fdc/files/5260309/rept.zip)

PeterBBCUser, 2020-09-22:
but if i capture it directly to .fsd it all messed up.
I am interested how or what did you use to convert it from rfi to fsd ?

picosonic, 2020-09-23:
I converted the RFI to FSD with the following command line

./bbcfdc-nopi -c -i Repton3-40tss0.rfi -l -summary -o rept.ssd

Only the bbcfdc-nopi executable can convert RFI files.

The "-l" option shows a layout map of where the sectors were recovered from on the disk surface.

The "-summary" option shows a summary which includes tracks with data, and the sector numbers, total recovered sector count, disk track range, drive tracks, single/double sided, FM/MFM/GCR/AppleGCR sectors found, detected density, sector size range, sector id range and total storage size in bytes.

I would recommend capturing in RFI in the first instance, then you can use the bbcfdc-nopi to convert to other imaging formats at a later time.

closed 2020-09-25 by PeterBBCUser