picosonic/bbc-fdc #26: Raspberry Pi Zero W

CommanderCoder, 2020-10-04:
Is it possible to get this to run correctly on Raspberry Pi Zero W.  I've use a RPi3 it works fine.  I switched to using a RPiZeroW and it gives errors.  Could it be a CPU frequency issue?

Great project and I've gone back to using the RPi3.

picosonic, 2020-10-04:
Hi,

  Can you send me the output of “cat /proc/cpuinfo“ on the ZeroW and a .RFI dump of the same disk as read by both the Pi3 and ZeroW, then I’ll take a look.

Make sure neither is over or under clocked.

Probably is cpu frequency as you say.

Thanks

Jasper

CommanderCoder, 2020-10-09:
pi@raspberrypi:~/bbc-fdc $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000003672ba4a

[Archive.zip](https://github.com/picosonic/bbc-fdc/files/5354826/Archive.zip)

picosonic, 2020-10-09:
Thanks for sending these over.

The .RFI files both say they were recorded at 7.8125 Mhz instead of the normal 12.5 Mhz

According to https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md your CPU revision code of 9000c1 corresponds to : 

Model : Zero W
Revision : 1.1
RAM : 512MB
Manufacturer : Sony UK
CPU : BCM2835

As far as I knew all BCM2835 boards run with a 250Mhz core, however the capture from the Pi Zero seems to suggest it's actually running with a 400Mhz (like a BCM2837).

This site seems to suggest that all Pi Zeroes run at 400MHz

https://www.raspberrypi.org/documentation/configuration/config-txt/overclocking.md

I've made a change to the code to assume Pi Zero is 400MHz now, can you download latest code please and test again.

Also I noted that in your text file the Pi3 dump says the code was compiled on a board with a BCM2835 processor, but the Pi3 has a BCM2837.

You need to make sure you compile the code on the Pi it is going to be run on.

Please do the following on each Pi before running on that hardware for the first time ..

make clean
make

CommanderCoder, 2020-10-11:
Thanks.  I did a rebuild and ran it, and it seems good.  I managed to capture images without error.

I captured two 'rfi' files from the same disk, one after the other, and the resulting files are different sizes.  Is that normal?  

I've attached the files
[Archive 2.zip](https://github.com/picosonic/bbc-fdc/files/5361494/Archive.2.zip)

picosonic, 2020-10-12:
Is this a 40 track disk in an 80 track drive?

When in auto detect mode, which is the default, it will try to read data to determine how the capture should run. However if it can't detect data reliably then the auto detect may fail.

In this instance your best bet is to do a full capture of everything by specifying the maximum number of tracks and sides to be captured to the RFI file

For example, this will capture both sides and all 80 tracks (assuming you have an 80 track drive) :

sudo ./bbcfdc -c -ds -tmax 80 -o diskimage.rfi

In terms of the resultant image files you attached, the smaller one was due to it detecting a single-sided 40 track disk in an 80 track drive and it employing double-stepping 

diskuser0787_rpi0_fixed.rfi
contains 40 tracks (0 to 39) single sided at 12.5MHz

diskuser0787_rpi0_fixed_a.rfi
contains 80 tracks (0 to 79) single sided at 12.5MHz

These captures are not as good as the previous RFI files you posted here, it looks like your heads need cleaning.

picosonic, 2020-10-15:
I'm closing this issue as the Pi Zero is now correctly detected and uses the right SPI clock divider based on the core frequency of 400MHz

closed 2020-10-15 by picosonic