picosonic/bbc-fdc #27: Head cleaning cycle [enhancement]

JasonStonier, 2020-10-06:
I've been thinking about the problem of cleaning the heads of a 5.25" drive - "spinning" cleaning discs are almost impossible to find these days, but I <think> you could clean the heads using an empty disc sleeve with a cleaning material and some IPA in the read window of the disc...if you could get the heads to track up and down a few times.

Presumably one could do this with something like [drivetest -tmax 40 -noindex -seek 40]

But, it seems to me (as a non-programmer) it could be fairly trivial to add a function to drivetest such as <drivetest -clean> which would run the heads to max and min a few times without spinning up the motor.

picosonic, 2020-10-06:
I've got a feeling that the heads won't load without the motor being on, so you can't do one without the other.
I'm pretty sure most drives won't step in and out without the motor going either.

I had a spinning cleaning disk (from the 80s) until quite recently, it was on a shelf in the server room in my office, and somebody chucked it out!

To make it do anything without a disk in the drive, you're right, you would need "-noindex"

Using "-tmax 40" will get it to step to 0, then to 39, and back again. It's used as a test to see how many tracks the drive supports seeking to.

The "-seek 40" would not be necessary unless you wanted it to seek back to track 39 after the "-tmax 40" command to leave the heads there.

Daniel Jameson recommended using a folded up lens tissue with a small amount of isopropyl alcohol on it. Place between the heads, close the drive door and gently move the lens tissue around. Be careful not to force the heads apart or the springs can be damaged.

JasonStonier, 2020-10-06:
I still wonder whether it might be worth looking at - I'll have a play with what I have using the tmax and seek parameters. I've made up a static cleaning disc using a sleeve from a dead disc and a coffee filter paper - the motor hole is left completely free so the motor can spin without driving anything.

JasonStonier, 2020-10-06:
It works, actually. Just tested it by chaining in a shell script 5x drivetest -tmax 40 -noindex

Cleaned the heads far better than I was managing with a Q-Tip. It may be that only I would ever be interested in this - in which case I'm quite happy to carry on doing it this way. Cheers!

picosonic, 2020-10-09:
I've added a "-clean" option which will attempt to step from track 0 to max track and back (without turning the motor on) 5 times.

I'm not sure if every drive will step without the motor enabled, but it's worth a try.

picosonic, 2020-10-15:
I'm closing this issue since the code was added, let me know how you get on

closed 2020-10-15 by picosonic