picosonic/bbc-fdc #29: Read and recovery data from DMF floppy disk (2HD 1.7MB)  - Enhancement

rataweb, 2021-03-16:
Hi,
I have old 2HD floppy disk formatted to 1.7MB (DMF). Please, could you add this feature (read tracks 81 and 82)?
Thanks,

picosonic, 2021-03-18:
Can you capture one of the disks in .RFI format and send it to me to check?

You can already capture more than 80 tracks using the -tmax parameter (remember tracks are 0 based)

When I looked up DMF format, it appeared to be a standard format, just with 21 sectors/track and 82 tracks.

I suggest you test your drive using drivetest to see what track number it is capable of stepping to

Thanks

picosonic, 2021-03-18:
I have found a Microsoft Windows 95 setup boot disk which claims to be in High-Density (DMF) format.

It has the following meta data

512 bytes/sector
1 sectors/cluster
1 reserved sector
224 root entries
2800 sectors on volume (small)
media type f0 [2.88MB 3.5-inch, 2-sided, 80-track, 36-sector or 1.44MB 3.5-inch, 2-sided, 80-track, 18-sector]
9 sectors/fat 
18 sectors/track
2 heads
0 hidden sectors
signature 29
volume id serial 2371-1907
volume label "NO NAME    "
system id "FAT12   "

WIN_OPK      28 W--VFA 00 09:58:30 31/03/1997
IO.SYS       20 W---FA 00 11:11:10 24/08/1996 @ 0x0002 -> 4200 214836 bytes
MSDOS.SYS    20 W---FA 00 11:11:10 24/08/1996 @ 0x01a6 -> 38a00 7 bytes
COMMAND.COM  20 W---FA 00 11:11:10 24/08/1996 @ 0x01a7 -> 38c00 93812 bytes
?AISATAP.SYS 20 W---FA 00 16:37:00 12/12/1996 @ 0x025f -> 4fc00 17961 bytes
autoexec.bat 20 W---FA 00 14:45:32 16/05/1997 @ 0x025f -> 4fc00 1362 bytes

bbc-fdc is able to extract the whole disk fine

18 x 512 byte sectors per track * 80 tracks on 2 sides = 1.44Mb so perhaps not really DMF

rataweb, 2021-03-19:
I'll try.
Thanks,

picosonic, 2022-01-07:
Hi, just following up on this, how did you get on?