picosonic/bbc-fdc #34: Unable to read 5.25 ADFS E discs

FrankHovis, 2023-09-02:
I have a number of 5.25 ADFS E format discs, non of which I can read. Ive successfully read 3.5 ADFS, 5.25 DFS and PC format discs, so I'm reasonably sure the hardware is ok.
ADFS E discs just don't get recognized:

frank@raspberrypi:~/bbc-fdc-master/tools $ sudo ./bbcfdc -c -v
Compiled on hardware with BCM2835 processor
Start with 937500 byte sample buffer
Starting at track zero
Disk is write-protected
Creating histogram for track 2, head 0 data sampled at 12500000 with 300.00 rpm
Maximum peak on track 2, head 0 at 42 samples, 3.360ms
  Peak at 42 3.360ms
  Peak at 61 4.880ms
  Peak at 81 6.480ms
Found 3 peaks

[45bd] FM Index Address Mark
[4bbe] GCR DATA

[6112] FM Index Address Mark
[9a8b] GCR DATA

[c123] FM ID Address Mark
[c35e] FM Track 252 (2) Head 127 (0) Sector 127 Data size 95 CRC df3f BAD (76c9)

[c860] FM ID Address Mark
[ca7c] FM Track 60 (2) Head 175 (0) Sector 255 Data size 255 CRC fdff BAD (0e8b)
[12340] GCR DATA
[16644] GCR DATA

[1ba5d] FM ID Address Mark
[1bc8a] FM Track 247 (2) Head 242 (0) Sector 255 Data size 239 CRC fb3f BAD (ad9e)

[1c69a] FM ID Address Mark
[1c8b7] FM Track 255 (2) Head 135 (0) Sector 115 Data size 247 CRC f7ff BAD (d669)
[1f410] GCR DATA

[1f50d] FM ID Address Mark
[1f72f] FM Track 95 (2) Head 255 (0) Sector 251 Data size 245 CRC 7fde BAD (ffc5)

[1ff1d] FM ID Address Mark
[20145] FM Track 244 (2) Head 191 (0) Sector 223 Data size 255 CRC dfff BAD (7d69)

[22811] FM ID Address Mark
[22a3d] FM Track 95 (2) Head 255 (0) Sector 255 Data size 255 CRC f7f6 BAD (924b)

[27acd] FM ID Address Mark
[27cef] FM Track 61 (2) Head 15 (0) Sector 255 Data size 255 CRC fffd BAD (c5a3)
[29727] GCR DATA

[298d3] FM Index Address Mark

[2b5fe] FM ID Address Mark
[2b834] FM Track 255 (2) Head 247 (0) Sector 247 Data size 255 CRC e7fa BAD (5835)

[2c143] FM ID Address Mark
[2c370] FM Track 254 (2) Head 31 (0) Sector 173 Data size 250 CRC b7df BAD (96c0)
[2d318] GCR DATA
[338f4] GCR DATA
[37683] GCR DATA

[3bdb2] FM ID Address Mark
[3bfd9] FM Track 255 (2) Head 221 (0) Sector 251 Data size 237 CRC eff8 BAD (6e2c)

[43b5b] FM Index Address Mark
[4415b] GCR DATA

[456af] FM Index Address Mark
[49027] GCR DATA

[4b6bd] FM ID Address Mark
[... 647 lines cut ...]
[af85c] GCR DATA

[b37a2] FM Data Address Mark, distance from ID b37a2
[b9132] GCR DATA
[da008] GCR DATA

Unknown logical disk format

Finished
Total extracted sectors: 0

Summary: 
Drive tracks 80
Single sided capture
Detected density : Unknown density 
Exit function

frank@raspberrypi:~/bbc-fdc-master/tools $ sudo ./drivetest
Start
Starting at track zero
Disk is write-protected
Approximate RPM 361.21
Exit function

Is ADFS-E Assuming 300 rpm? Can I force a format to be used?

picosonic, 2023-09-02:
Can you capture disk to a .RFI file and send to me to have a look at please?

The RPM mismatch could well have an effect, as it is taken into account when calculating bit cell window size.

drivetest will count index pulses on track zero to determine RPM, the capture program should do the same on a per track basis, but looking at the code it only does this when writing to .SCP and .RFI files.

300 RPM is the default unless a different RPM is detected.

You could try capturing to RFI file and then use that capture file as input when converting to an ADFS disk format. That way the RPM would be captured correctly.

Let me know if this works and I can update the code.

FrankHovis, 2023-09-02:
[ADFS L format.zip](https://github.com/picosonic/bbc-fdc/files/12504732/ADFS.L.format.zip)

I've tried various attempts with different discs all in ADFS-L format, using various different tried with sides/sectors etc. I just can't seem to get anything to work. I'm starting to wonder if I have some hardware issue as:

_Single-sided disk assumed, only found data on side 0_

doesn't seem at all right, some of these discs should be full. 

I have two drives, both Canon MD5501.  I can read both sides of a BBC DFS disc, but it'd be just my luck for them to both be faulty.

picosonic, 2023-09-03:
I've managed to get a catalogue from that image by manually overriding the bitcell window size down to 3.33us (from 4us).

I also got sector data out of the file by converting the RFI to SCP and loading it into HxCFloppyEmulator, then dumping to generic IMG format, so that may be something you could try.

Detected ADFS-L
Disc name: "BBCdisc1  "
Disc size in (256 byte) sectors: 2560
Disc ID: d344 (54084)
Boot option: 03 *Exec boot file

!BOOT      R-LF--         48 000234 00000000 ffffffff
!MAIN_MENU R-LF--        462 0002b0
BBCROMS    R-LD--       1280 000025
  !Info      RW-F--       2637 000850 00000000 ffffffff
  ASSEMBLER  RW-F--      16384 000410
  Command    RW-F--      16384 000450
  Edit       RW-F--      16384 000490
  ETROM      RW-F--       5376 0004d0
  HLPROM     RW-F--      16384 0004e5
  Kermit     RW-F--      16384 000525
  MEX        RW-F--      16384 000565
  RAMPAD     RW-F--      16384 0005a5
  SLAVE      RW-F--      16384 0005e5
  Sp8000     RW-F--       8192 000626
  SYMDIS     RW-F--       5152 000646
ELKINTRO   R-LD--       1280 000177
  !Info      RW-F--        182 000033 00000000 ffffffff
  !M1        RW-F--       1493 000661
  BIORHYTHMS RW-F--       4184 00017c
  BUGZAP     RW-F--       4109 00018d
  BYE        RW-F--        104 0001a3
  CLOCK      RW-F--       2824 0001a4
  DODGEMS    RW-F--       4128 0001b0
  GOMOKU     RW-F--       6560 0001c6 00000e00 00008023
  GREETER    RW-F--       1108 0001e0
  INTRO      RW-F--       4440 0001e5
  ISLAND     RW-F--       1217 0001c1
  ISLESKR    RW-F--      14277 0001fc 00000e00 00008023
  KEYBOARD   RW-F--      14038 000235
  MARSLANDER RW-F--       5431 00026c
  MESSAGE    RW-F--       1593 000282 00000e00 00008023
  PATTERNS   RW-F--       2473 000289 00000e00 00008023
  PIANO      RW-F--       6002 000293
  PLANETS    RW-F--       1240 00019e
  PLASKR     RW-F--      11159 0002b2 00000e00 00008023
  RIVERGAME  RW-F--       4980 0002de
  SKETCH     RW-F--       3527 0002f2
  SOLVER     RW-F--       9341 00031b 00000e00 00008023
  TURTLE     RW-F--       6039 000340
M          R-LF--       1348 00065b
MASTER     R-LD--       1280 000007
  !Info      R-LF--        350 0000c4 00000000 ffffffff
  !M1        RW-F--       1053 0001f7
  aaobj      R-LF--       6205 00000c 00002fbd 00002fbd
  Adventure  R-LF--      10948 000034 00000e00 0000802b
  Aqua       R-LF--        965 00005f 00000e00 0000802b
  Attack     R-LF--       1076 000063 00002b00 00002b00
  Castle     R-LF--       4219 000068 00000800 0000802b
  CharDes    R-LF--       5306 000079 00000800 0000802b
  Cloud      R-LF--       2113 00008e 00000800 0000802b
  Clown      R-LF--       2454 000097 00000e00 0000802b
  Convert    R-LF--       1367 0000a1 00000e00 00000e00
  Envelope   R-LF--       4631 0000a7 00000800 0000802b
  FONTS      R-LF--       2305 0000ba 00002000 00002000
  Keyboard   R-LF--       7832 0000c6 00000e00 0000802b
  Modes      R-LF--       7413 0000ea 00000e00 0000802b
  Panel      R-LF--       5887 000107 000018bd 000018bd
  Patterns   R-LF--       1600 00011e 00000e00 0000802b
  Pfill      R-LF--       4335 000125 00000800 0000802b
  PFONT      R-LF--        171 000136 000023e1 000023e1
  Shapes     R-LF--       1670 000137 00000e00 0000802b
  TimPaint   R-LF--       7892 00013e 00000e00 0000802b
  Turtle     R-LF--       3497 00015d 00000e00 0000802b
  Welcome    R-LF--       3037 00016b 00000800 0000802b
MUSICAL    R-LD--       1280 000358
  !Info      RW-F--        170 000625 00000000 ffffffff
  !M1        RW-F--       1114 0002ab
  +BlueSky   RW-F--       3456 0003c0
  +JukeTn1   RW-F--       3584 0003b2
  Aha        RW-F--       3708 000797
  Arpegiator RW-F--       1195 0008d4
  ClodEat    RW-F--       1430 000747
  Doctor     RW-F--       9659 00074d
  Joplin     RW-F--      19456 00035d
  JukeBox    RW-F--       2176 0003a9
  Music      RW-F--       4322 000773
  POSTMAN    RW-F--       1386 0007a6
  PYTHON     RW-F--       3945 0003de
  RENDAVOUS  RW-F--       4074 0003ce
VARIOUS    R-LD--       1280 0003ee
  !Info      R-LF--        194 000031 00000000 ffffffff
  !M1        R-LF--       1244 000300
  7card      R-LF--       4029 00089a
  Animate    R-LF--       1664 00002a
  AUmaze     R-LF--       5892 0003f3
  Bus        R-LF--       3002 0007b7
  Clock      R-LF--       3029 00082b
  Copy       R-LF--        978 000837
  Game       R-LF--      23552 000667 00002000 00002000
  Giz        R-LF--       6656 0006c3
  GorbFruit  R-LF--       4768 000784
  Guide      R-LF--       5024 0006dd
  Hedgehog   R-LF--       5169 00083b
  Henge      R-LF--       3507 0007c3
  Hexmon     R-LF--       1159 0000e5
  Imagen     R-LF--       3476 0006f1
  K          R-LF--      10659 0008aa 00000000 ffffffff
  LABYRINTH  R-LF--       1075 000826 00004254 00004254
  Labyrinth1 R-LF--      21539 0007d1 00001100 00005faa
  Micky      R-LF--       1122 00040b
  MIM        R-LF--       5116 00071d 00000f00 00008023
  Moon       R-LF--        830 00085b
  Munch-Man  R-LF--       5545 000305
  PB         RW-F--       6417 000866
  Pelman     R-LF--       1672 00085f
  Pelman2    R-LF--       3854 00070d
  PI         RW-F--        256 000032 00000a00 00000a00
  PINBALL    RW-F--       6405 000880
  Pola       R-LF--       3428 0006ff
  Sexi       R-LF--       5504 000731
  Surface    R-LF--       2816 0007ac

paulrho, 2023-09-03:
Just a guess: could this be a 360 RPM instead of 300 RPM issue?Sent from my iPhone

picosonic, 2023-09-13:
I've made a number of changes for this.

Firstly you can force an RPM value on the command line, this is incase the RPM was calculated wrong or the index sensor on the drive isn't working correctly.

Next I've fixed the bitcell window adjustments based on the RPM variance from target RPM.

Lastly I've changed the DFS detection algorithm to only validate against disks encoded in FM format, this is because the RFI you posted was validating as DFS even through it was ADFS.

I'll close this issue for now as it's working for me with your RFI image. Any further issue please raise a new issue on here.

Thanks

closed 2023-09-13 by picosonic