picosonic/bbc-fdc #35: CPU / processor identification in hw.sh during compilation on later Raspberry Pi OS versions.

MindrustUK, 2024-04-24:
hw.sh no longer returns an expected string on later versions of Raspberry Pi OS

Building on a fresh Raspberry Pi OS install on a Raspberry Pi 3 and getting a few "Built on unknown processor" warnings. I looked into hw.sh and how it works. The expected output from `cat /proc/cpuinfo`  is not as expected and always ends up with `HW_UNKNOWN`.

Here is a sample of what is returned:

`processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Revision        : a02082
Serial          : REDACTED
Model           : Raspberry Pi 3 Model B Rev 1.2
`
This lead me to the following documentation: [https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/revision-codes.adoc)

It seems the way to go is to read information from elsewhere and set the CPU accordingly through a lookup table or similar.

Hope this is a useful report.

MindrustUK, 2024-04-25:
Operating system was written with the Linux variant of the Raspberry Pi Imager tool - Default selection "Raspberry Pi OS (64-bit) 2024-03-15"

Output from shell:

cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Using the latest version of the code from this repo `git clone https://github.com/picosonic/bbc-fdc` as of ~1 minute ago.

Output of ./hw_all.sh:

./hw_all.sh 
HW_UNKNOWN
REV_a02082
RAM_1024
MAN_SONY_UK
CPU_BCM2837
BOARD_3B
BOARDREV_2

Tested as my local user and as root with the same result.

picosonic, 2024-04-25:
Ok, thanks for the reply.

The hardware setting is only used to report when the program is run what hardware it was compiled on, which is only an indicator when running the tool incase the code is compiled on a machine with a different clock and therefore SPI speed to the machine it's being run on. i.e. compiled on RPI3 but run on RPI2 would not give good results.

The revision code is what is used to determine the actual hardware and set up the clock. Which is in turn used to work out the SPI divider values and buffer sizes.

So for your revision code of a02082, the clock is set as 400.

```
// Try to detect RPi 3 / 3A+ / 3B+
#if defined(REV_9020e0) || defined (REV_a02082) || defined(REV_a020a0) || defined(REV_a020d3) || defined(REV_a22082) || defined(REV_a220a0) || defined(REV_a32082) || defined(REV_a52082) || defined(REV_a22083) || defined(REV_a02100)
  #define RPI3 1
  #define HAS_BCM2837 1
  #define CLOCK_400 1
#endif
```
The program will still work in the same way, despite not being able to get the hardware type from /proc/cpuinfo

Maybe an improvement would be to ditch the hardware value and instead calculate it from the revision code.

Thanks

MindrustUK, 2024-04-25:
Good to know, I manually set the CPU value to just get things going without warning.

Thanks for the work! Feel free to close or leave as a note for future improvement as you see fit.

picosonic, 2024-04-25:
My Pi3 B with a revision code of a020d3, running on "Raspbian GNU/Linux 11 (bullseye)" gives ..

```
Hardware	: BCM2835
Revision	: a020d3
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
```
Which decodes as :

```
--------101000000010000011010011
NOQuuuWuFMMMCCCCPPPPTTTTTTTTRRRR

F = new style revision
MMM = 1GB
CCCC = Sony UK
PPPP = BCM2837
TTTTTTTT = 3B+
RRRR = (1.)3
```

So the Hardware value returned here of **BCM2835** is wrong, it should be reporting **BCM2837**

This is might be why they took it out of /proc/cpuinfo.