picosonic/bbc-fdc #36: binary

jrweikle, 2024-06-01:
is there a precompiled binary available?

or directions on making the binary for someone who has never done so before?

jrweikle, 2024-06-17:
ok finally figured out how to make the binaries.  The software says it doesn't see the drive index.   what are the jumper settings suppose to be on?

picosonic, 2024-11-01:
Hi, the jumpers are for switching between 5-inch and 8-inch drives

J2 switches between using index sensor on pins 4 (8 inch drive) and 8 (5 inch drive) of the shugart cable
J3 switches between drive 0 and drive 1 select
J4 switches between side (head) select on pins 32 (5 inch drive) and 2 (8 inch drive)

hope that's helpful.

I use it in the 5 inch position, then only normally move the drive select jumper depending on what drive I connect. 3.25 inch PC drives tend to favour drive 1, older ones are more likely to default to drive 0, but it depends if your shugart cable has a twist and if it does which side of the twist you connect your drive.

This is why I created the drivetest program, to hopefully allow you to verify your wiring and signals

jrweikle, 2024-11-01:
thanks i will try that some time this weekend

closed 2025-03-03 by picosonic